arpruss/ADCTouchSensor #2: Missing function?

rlogiacco, 2018-09-26:
While trying to compile for STM32 architecture I got a compilation error at https://github.com/arpruss/ADCTouchSensor/blob/b0ebd814f01e05ce39d4d53f2ad145412f8c3a92/src/ADCTouchSensor.cpp#L79

```
ADCTouchSensor\0.0.9\src\ADCTouchSensor.cpp: In member function 'int ADCTouchSensor::readRaw(unsigned int)':
ADCTouchSensor\0.0.9\src\ADCTouchSensor.cpp:79:20: error: 'ground' was not declared in this scope
             ground();
                    ^
```

Am I missing something?

203Null, 2019-04-28:
It only works on f103c, you can manually select the grounded pin for other stm32, or find a grounded pin of  the ADC.

ChoCho59, 2020-07-17:
I face the same problem. 

I don't know how to "manually select the grounded pin for other stm32, or find a grounded pin of the ADC"  !

arpruss, 2020-07-17:
I fixed the bug in the code.

In the constructor, you need to use sacrificialPin to specify an analog pin that you won't use for touch sensing or anything else but which will be used for grounding the internal ADC capacitor. You then leave this pin floating in your project.

closed 2020-11-11 by arpruss