arpruss/ADCTouchSensor #7: Error compiling for WeAct Blue Pill-Plus STM32F103

Dans-Droids, 2021-12-05:
Hi wonderful people, I'm having the following error popping up in the Arduino IDE during compilation..

....\Documents\Arduino\libraries\ADCTouchSensor-master\src/ADCTouchSensor.h:46:87: error: 'ADCTOUCH_DEFAULT_DELAY' was not declared in this scope
   46 |         ADCTouchSensor(int pin, int sacrificialPin=-1, unsigned delayTimeMicroseconds=ADCTOUCH_DEFAULT_DELAY);
      | 

I have tried accounting for this by including various forms of:

ADCTouchSensor ADCTOUCH_DEFAULT_DELAY = 500;

but the error persists.. 

Does anyone know a workaround?!
Thank you!

arpruss, 2021-12-05:
Which core are you using?

Dans-Droids, 2021-12-05:
Thank you for the response, however I am unsure as how to answer.. 
To which core do you refer?

arpruss, 2021-12-05:
The Arduino IDE core. Is it libmaple?

Dans-Droids, 2021-12-05:
Ah my bad, I believe it is yes - I did not flash the STM32 myself so I'm not 100%..

arpruss, 2021-12-06:
What does the Arduino IDE show as the board?

Dans-Droids, 2021-12-06:
the IDE is set to:
Generic STM32F1 series, BluePill F103C8, USART Enabled, USB:CDC, USB SPEED: LOW/FULL, C RUNTIME: Newlib Nano, Upload Method: HID BOOTLOADER 2.2

arpruss, 2021-12-06:
Try just defining ARDUINO_ARCH_STM32F1 in the library header file and see if it works.

Dans-Droids, 2021-12-06:
Thank you for the suggestion, I have tried a few different things in the header file but  all still give the same error.. here is a full print out of the error:

WARNING: library ADCTouchSensor-master claims to run on STM32F1, STM32F2, STM32F3, STM32F4, avr, STM32 architecture(s) and may be incompatible with your current board which runs on stm32 architecture(s).
In file included from C:\DansDroids Main\TR570 - Nextion Tricorder\Nextion_STM32_Tricorder\Nextion_STM32_Tricorder.ino:15:
C:\Documents\Arduino\libraries\ADCTouchSensor-master\src/ADCTouchSensor.h:46:87: error: 'ADCTOUCH_DEFAULT_DELAY' was not declared in this scope
   46 |         ADCTouchSensor(int pin, int sacrificialPin=-1, unsigned delayTimeMicroseconds=ADCTOUCH_DEFAULT_DELAY);
      |                                                                                       ^~~~~~~~~~~~~~~~~~~~~~
exit status 1
Error compiling for board Generic STM32F1 series.

I noticed that the message at the top is saying I am using "stm32" rather than "STM32"??

arpruss, 2021-12-07:
Did you define ARDUINO_ARCH_STM32F1 at the top of the header?

Dans-Droids, 2021-12-07:
I did, however with it as the first line, this is the error given:

WARNING: library ADCTouchSensor-master claims to run on STM32F1, STM32F2, STM32F3, STM32F4, avr, STM32 architecture(s) and may be incompatible with your current board which runs on stm32 architecture(s).
C:\Documents\Arduino\libraries\ADCTouchSensor-master\src\ADCTouchSensor.cpp: In member function 'void ADCTouchSensor::ground()':
C:\Documents\Arduino\libraries\ADCTouchSensor-master\src\ADCTouchSensor.cpp:49:5: error: 'adc_read' was not declared in this scope
   49 |     adc_read(ADCTOUCH_STM32_ADC, ADCTOUCH_STM32_GROUND_CHANNEL);
      |     ^~~~~~~~
exit status 1
Error compiling for board Generic STM32F1 series.

arpruss, 2021-12-07:
I think you're using the official STM32 core, and you should use the libmaple one. Here are some instructions: https://www.instructables.com/Programming-STM32-Blue-Pill-Via-USB/

Dans-Droids, 2021-12-07:
Thank you for the link however i dont think this will be compatible with my board as it is not a generic STM32, it is made by WeAct and called "Blue Pill-Plus"..

WeAct provide a customised bootloader in two flavours, one is a modified version of SMT32duino.. 

this is the board:    [https://github.com/WeActTC/BluePill-Plus](https://github.com/WeActTC/BluePill-Plus)

arpruss, 2021-12-07:
If it's the same processor, it should work.

Dans-Droids, 2021-12-09:
hey, so I've been playing around the last couple of days and found a way around the error, by adding a || defined(GENERIC_stm32) to the end line of the initial #if defined(ARDUINO_ARCH_STM32F1)......
and adding:

ADCTouchSensor(int pin, int sacrificialPin=-1, unsigned delayTimeMicroseconds=20);     <-- this delay of 20 (taken from the above mentioned definition) 

however I now get this error:

WARNING: library ADCTouchSensor-master claims to run on STM32F1, STM32F2, STM32F3, STM32F4, avr, STM32 architecture(s) and may be incompatible with your current board which runs on stm32 architecture(s).
C:\Users\dan\OneDrive\Documents\Arduino\libraries\ADCTouchSensor-master\src\ADCTouchSensor.cpp: In member function 'int ADCTouchSensor::readRaw(unsigned int)':
C:\Users\dan\OneDrive\Documents\Arduino\libraries\ADCTouchSensor-master\src\ADCTouchSensor.cpp:86:20: error: 'ADCTOUCH_DIVIDER' was not declared in this scope
   86 |     return total / ADCTOUCH_DIVIDER / samples;
        |                    ^~~~~~~~~~~~~~~~

Dans-Droids, 2021-12-11:
I was able to upload the sketch but the returned value is always -10000?

arpruss, 2021-12-11:
Why don't you try it with the libmaple core?

Dans-Droids, 2021-12-11:
I did try but nothing on the board worked, there was no serial output in the Arduino IDE either so I re-flashed the one from WeAct and everything worked again.. Maybe the WeAct boards just arent compatible with this library? 

Thank you for your help, I will just bite the bullet and buy an MPR121 breakout board, a little overkill for just 1 capacitive sensor lol

arpruss, 2021-12-11:
Did you flash a new bootloader using a USB serial device when you used the libmaple core?

Dans-Droids, 2021-12-11:
yes I used the ST-link and STCubeProgrammer to flash both

arpruss, 2021-12-11:
Can you get the LED to blink at least?

Dans-Droids, 2021-12-11:
no, I could not get the LED to blink on the standard STM32duino bootloader and libmaple core, even when declaring the pin for it in the sketch.. I think part of the issue is that with my board, the LED is on PB2 and other things are not tied to the same pins as they are on a generic smt32f1..

arpruss, 2021-12-11:
I don't know. How about just getting a normal $2+shipping blue pill ?

closed 2021-12-11 by arpruss

Dans-Droids, 2021-12-11:
hahaha, good idea.... I did try it on my other STM32F103C8 board, which uses exactly the same compilation settings but is flashed with the same bootloader linked in that YT video (presuming it is libmaple) and it gave the same response..

arpruss, 2021-12-11:
Did you install the libmaple core in Arduino from Roger's GitHub, and what board settings did you use in Arduino?