arpruss/ADCTouchSensor #11: Clarification needed on internal grounded ADC channel

CrouchingPanda, 2024-07-17:
I am trying to see if below statement applies to my STM32F072RB,
> On the stm32f103c, you can omit the sacrificial pin, and an internal grounded ADC channel can be used. If your STM32F1 MCU has such a channel -- likely, the same one, namely 15 -- you can edit the library to support it (and email me).

Could you advise where in the documentation (datasheet or maybe other documents) should I look for this data?

arpruss, 2024-07-18:
It's too long ago, so I don't know.

Looking at the datasheet, the stm32f103c only has 10 usable ADC channels. I am guessing that the remaining channels are still available but are grounded, and that's what I'm using.

On your chip, it may be that all 16 channels are usable, in which case you might not have an always-grounded channel, and you'll have to sacrifice one of the 16 to ground.

CrouchingPanda, 2024-07-21:
Thank you.

Based on what is described in README, this sacrificial pin shouldn't be connected to anything. On my PCB unused pins are grounded through a resistor. So then should I ground the sacrifical pin like that as well (externally), or should I leave it fully unconnected?